tum-esm/hermes #9: Add activity chart

empicano, 2022-10-30:
What do you think about adding an activity chart, showing sensor activity over the last month or so? I implemented this for the AQSN project, like so:

[image]

If there's interest, I'll implement a similar query here and return the activity data together with the sensor configurations on the `GET /sensors` route.

The dashboard can follow whenever, I'm only asking because it would fit well for me atm.

dostuffthatmatters, 2023-05-23:
I would like to add this :) which HTTP route should I use to get the aggregated heartbeats. Very low priority, though.

patrickjaigner, 2023-10-17:
Dashboard is developed further in different repository.

closed 2023-10-17 by patrickjaigner