tum-esm/hermes #11: Run GitHub Workflows conditionally

dostuffthatmatters, 2022-11-03:
TODO

empicano, 2022-11-07:
For the scope naming, I'd go with what Karma specifies:

> The `<scope>` can be empty (e.g. if the change is a global or difficult to assign to a single component), in which case the parentheses are omitted. In smaller projects such as Karma plugins, the `<scope>` is empty.

dostuffthatmatters, 2022-11-09:
Who is Karma?

dostuffthatmatters, 2022-11-09:
Aaaah 🫡

I thought you were contributing to the issue in some way 😉😅

empicano, 2022-11-09:
Of course not.

If case you _were_ looking for meaningful contributions: We don't need to hack around with hashes. See https://docs.github.com/en/actions/using-workflows/workflow-syntax-for-github-actions#onpushpull_requestpull_request_targetpathspaths-ignore

closed 2022-11-09 by dostuffthatmatters