tum-esm/hermes #39: Route to read all logs

dostuffthatmatters, 2023-02-06:
It would be very helpful to have a route to read all log messages from one sensor. The log aggregation does not contain details or info messages.

Like the `GET /sensors/<sensor-identifier>/measurements` route but for logs: `GET /sensors/<sensor-identifier>/logs`.

dostuffthatmatters, 2023-04-14:
Could this be implemented soon?

empicano, 2023-04-28:
Note that the paging logic changed slightly:

```
The query parameter `creation_timestamp` defines the base timestamp for the query.
Depending on the value of the query parameter `direction`, the request returns
the "next" or "previous" N records. If `direction` is not set, it defaults to
"next". If `creation_timestamp` is not set, the request returns the first N records
if `direction` is "next" and the latest N records if `direction` is "previous".
```

In particular, this breaks the current dashboard because it now fetches the _first_ N measurements instead of the _latest_ N measurements when no query parameters are passed. Add the query parameter `?direction=next` to get the latest N measurements.

closed 2023-04-28 by empicano

dostuffthatmatters, 2023-04-28:
Nice! Thank you for implementing, it :) Will fix the dashboard next week.

commit b997cae5 (2023-04-28): feat: Add route to page through logs; Close #39