tum-esm/hermes #86: Heartbeat Messages returned from `logs` route

dostuffthatmatters, 2023-05-23:
@empicano Is that how it is supposed to be. I am pretty sure, that I am sending the heartbeats to the correct topic.

[image]

empicano, 2023-06-05:
Yes, this looks good 👍

I'm writing a log entry with severity `system` to the database when the server receives a heartbeat. The idea is to have these events as basic logs even when the sensors don't implement any logging. Not sure yet if this is a good idea.

dostuffthatmatters, 2023-06-05:
Ah, ok great!

I think it is a great idea to sort the heartbeat messages into the log stream. I am a bit unsure about the "system" label, but I have not strong opinion or counterproposal 😅 

What makes the log page a bit tricky now, is that when I now fetch the raw logs for one sensor, I will only fetch heartbeat messages for the first x pages. How should I deal with that if I don't only want to display heartbeat messages?

empicano, 2023-06-05:
Hm, good point. The best option would probably be to implement full-text search over the logs. Filtering by log level would probably suffice in the beginning. That's a lot of work though.

Another option would be to stay with a simple way to show the logs but be able to stream them over to a dedicated log service (like DigitalOcean does with its App Platform). I lean towards this option.

dostuffthatmatters, 2023-06-05:
> Hm, good point. The best option would probably be to implement full-text search over the logs. Filtering by log level would probably suffice in the beginning. That's a lot of work though.

As a quick fix, could you hide the system logs from the raw logs returned by the server? We still have the heartbeat activity via the aggregation, for the activity plots, right?

empicano, 2023-06-06:
I no longer write the messages to the sensor logs now. On thinking about this, it's probably also cleaner to keep the sensor logs to the sensors and not mess with them server-side at all.

The aggregation currently runs only over the measurements, aggregating over all received messages could be something interesting in the future, but not a priority right now I think. Especially because that aggregation is not trivial.

My idea with the heartbeats is to have a consistent connection metric to see if the sensor is online/unstable/offline without relying on parsing the logs. I still need to think about how to implement this in the database.

closed 2023-06-06 by empicano

dostuffthatmatters, 2023-06-06:
Ok, good, thank you! :)

We can talk about ideas how to use the heartbeats in another way.

empicano, 2023-06-06:
Yes, always open to ideas! 😊

patrickjaigner, 2023-06-07:
I hope I understand the goal of the current logic correctly. An idea:

At my old company, we had a heartbeat logic that would only send a heartbeat ("keep-alive") if no other message was sent for at least 30 seconds. Therefore you can expect a baseline communication over the MQTT topic. If nothing arrives you know that something is wrong. If there is at least 1 message per Minute you knew that at least the connection is stable. 

1. This would at least reduce heartbeats to way less traffic
2. They are only sent if nothing else is happening

empicano, 2023-06-07:
That's some good insights 👍 The traffic from heartbeats is not really a problem from a technical standpoint (so 1. & 2. are not that severe), I prefer to look at it from a usability standpoint and go from there: how does it help us make decisions and how do we then best represent the metric on the dashboard?

I tend towards a combination of:

- Simple online/unstable/offline metric that just shows if we received a heartbeat in some recent time interval
- When did we receive the last heartbeat?

This would help us easily check if a sensor is connected, and when the problem occurred if the sensor loses the connection

---

Calculating the metric on the server is definitely an option, but I would still prefer to have the heartbeats in the database. Persisting data over longer durations on the server is not what it's made for (and not necessarily less complex).

The first bullet point would still work like this, but the second one would fail when the server restarts, because the queue is emptied, and thus we can't know when we received the last heartbeat of a sensor currently without connection.

commit cf646125 (2023-06-06): refactor(server): Stop writing to sensor logs server-side; Close #86