tum-esm/hermes #87: Missing in-flow data during calibration

patrickjaigner, 2023-05-26:
There is no in-flow sensor data being communicated during calibration.

TODO: Check if the data is logged locally.

Some examples plotted from the database:
[image]
[image]

dostuffthatmatters, 2023-05-26:
No, the data is not logged locally. This data is only logged every measurement iteration during regular measurements.

patrickjaigner, 2023-05-26:
There is a high priority to have this data for proper analysis during calibration and system analysis.

patrickjaigner, 2023-06-27:
Merged in https://github.com/tum-esm/hermes/pull/101

closed 2023-06-27 by patrickjaigner