tum-esm/hermes #88: Upgrade Systems to Beta 2

dostuffthatmatters, 2023-06-06:
- [ ] 1
- [ ] 2
- [ ] 3
- [ ] 4
- [ ] 5
- [ ] 6 (not possible yet)
- [ ] 7
- [ ] 8
- [ ] 9
- [ ] 10
- [ ] 11 (not possible yet)
- [ ] 12 (not possible yet)
- [ ] 13 (not possible yet)
- [ ] 14 (not possible yet)
- [ ] 15 (not possible yet)
- [ ] 16 (not possible yet)
- [ ] 17 (not possible yet)
- [ ] 18 (not possible yet)
- [ ] 19 (not possible yet)
- [ ] 20 (not possible yet)

dostuffthatmatters, 2023-06-12:
> @dostuffthatmatters I would like to start working on 1-2 additional systems. Can you get some ready today?

Hi! Yes, I can get some ready after lunch. Will ping you then.

dostuffthatmatters, 2023-06-12:
@patrickjaigner `8`, `9` and `10` are good to go. The battery of `10` seems very weak.

patrickjaigner, 2023-06-12:
@dostuffthatmatters thanks for the update. Keep system 10 powered by the grid then. The UPS will take care of loading the battery.

dostuffthatmatters, 2023-06-19:
I update a few more systems. 17 possibly has a broken battery (10 minute charging, off after 2 seconds).

patrickjaigner, 2023-06-19:
I fixed system 3 this afternoon. It should be up and running again.