tum-esm/hermes #90: Server not receiving all log messages?

dostuffthatmatters, 2023-06-07:
If you look at the aggregated logs of sensor 1, the last log message (except heartbeats) is 45 days old:

[image]

There are significantly newer log messages than that. @empicano could you check, whether more log messages have arrived since then?

dostuffthatmatters, 2023-06-07:
> I can't find newer logs in the database. Locally, receiving logs works fine and I don't see any errors in the server logs either.

Weird.

> Is there a sensor actively sending logs at the moment so that we can test this?

We can test this during the Friday meeting.

patrickjaigner, 2023-10-17:
Root cause: MQTT topics on the system and server where slightly different.

closed 2023-10-17 by patrickjaigner